mzbotreprap/VORON #1: Documentation PDF is corrupted as of 3/10

kevjustice, 2016-03-10:
Tried to view it but after about 5 pages it cuts in the middle (even viewing online within GIT).

vonnieda, 2016-03-10:
@kevjustice May be your browser or reader. Working fine for me. I get 7 pages with a "Congrats" message at the end both in Chrome within Github and downloaded and opened in Preview.

mzbotreprap, 2016-03-10:
Do you have the same issue when you "git clone" and open locally?

kevjustice, 2016-03-10:
Maybe it's a browser/GIT issue.  When I download the entire zip it's fine.  But downloading just the pdf and Adobe won't even open it and in the GIT viewer it shows like the image attached.  At least I got it.
[image]

closed 2016-03-10 by mzbotreprap