mzbotreprap/VORON #6: Screw-set

Printeronimo, 2016-03-31:
Hey Maks,
I really LOVE your design. I want to build a corexy machine and was struggling between the Sparkcube and the Smartcore/Smartrapcore
Sparkcube is too expensive for me and Smartrapcore is really bad documented! 
I think your documentation as well as the build-manual are killer and therefore decided to build the voron.
you mentioned on some website that you´d sell a screw-kit for 32 bucks. This would save me a lot of time! would you also send it to germany? i could transfer you the money via Paypal if you tell me the shipping costs.

kind regards,

PS. sorry for the raised issue. i did not figure out how to communicate on Github and i deleted my Facebook account several years ago.

Michael
likethespike(at)gmail(dot)com

mzbotreprap, 2016-03-31:
Gonna take this offline :)

mzbotreprap, 2016-04-01:
Or not :)

Hey Michael. I emailed you at the address you provided in the signature. I'm migrating DNS right now so my email has been spotty. 

GitHub has removed messaging feature a while ago. There's a hack to get it to work, but it's really more work than it's worth. Feel free to email me at mzolin(at)mzbot(dot)us if my email never arrived.

mzbotreprap, 2016-04-02:
Houston, we have contact. Closing.

closed 2016-04-02 by mzbotreprap