mzbotreprap/VORON #7: Locating BOM Stepper motor wire extension cable (1M)

codexmas, 2016-04-01:
Searching Aliexpress for variants of this doesn't yield any sane results.
Can you please post a link to the product you found?

I am also considering just building them myself, but for $4/ea I am inclined to buy them if they are of good quality.

mzbotreprap, 2016-04-02:
Fun fact: If you are not planning to have an LCD panel, you have safely mount the controller (RAMPs or otherwise) in the back, next to the PSU. Then you no longer need these extensions, since the BOM specifies motors that have sufficient length wire to reach it.

I'm not sure why I put 1M length requirement. I believe this was before I found steppers with a meter of cable attached. You really only need extra 300mm. Those can be had here: http://www.ebay.com/itm/10pcs-30cm-4Pin-Male-to-female-Jumper-Wire-Dupont-Cable-for-3D-Printer-/281763046598?hash=item419a62e4c6:g:T20AAOSwu4BVvHWV

I've seen listings for less than 10pcs, since you really only need 1. The B motor wire is not quite long enough to reach the controller if it's on the front.

And this investigation lead me to find that the stepped part numbers are wrong again. The parts you want (one with 1M leads) is 17HS19-2004S. I'll update the BOM...

mzbotreprap, 2016-04-02:
BOM updated. Closing.

closed 2016-04-02 by mzbotreprap