mzbotreprap/VORON #9: BOM Consider adding the Torx tapping screws to the hardware kit

codexmas, 2016-04-01:
If these aren't included in the kit, please consider adding them to the kit, to reduce the number of small parts required. With that change all the fasteners would be in one kit, making assembly a bit simpler.

mzbotreprap, 2016-04-01:
This is a tricky one. The reason I didn't include them is they are expensive. $15 for a set of 24.

I wanted people to have options for using normal low profile M5 screws and tapping the extrusions. It's a huge pain in the ass, but it's cheaper.

To me, $15 is worth not having to tap 12 extrusions on each end :) Thoughts?

codexmas, 2016-04-01:
Yeah, I just ordered the set and they are expensive, but worth it.
Maybe 2 kits? One with the Torx, and one with the low profile M5 screws and possibly with a tap?
Possibly going too far, as I know it can be a pain to assemble the kits.

I personally have a whole set of metric and imperial taps, but I think that with the quality of the documentation and the level of printer you end up with it will appeal to a much larger base.

mzbotreprap, 2016-04-02:
So, low profile M5 screws and a decent tap will not be much cheaper than the thread-forming screws. Screw it (pun intended), I'm adding them to the kit.

closed 2016-04-02 by mzbotreprap