mzbotreprap/VORON #11: Greasing method recommendation

dgcaste, 2016-04-10:
linear bearing greasing

I'm going to recommend a faster and more effective method of greasing. It also does a very good job of covering all the nooks and crannies, pushing out dirt, and squeezing  out the transport oil included with the bearings.

1.	The best time to do this is after the bearing has been placed in its housing because it makes it easier to keep control of the bearing.

2.	Set up on a lint-free rag, such as a clean and new white t-shirt, or get lint free rags from the hardware store. Gloves are recommended to minimize contamination, and you'll be glad you can take them off and not be covered in grease.

3.	Clean two smooth rods thoroughly with another (or the same) rag and set them down on the clean area.

4.	Place a tube of grease flush with the bearing opening, then squeeze the tube until about 50% of the bearing is full of grease.

5.	Push a smooth rod through the grease plug and watch it come out the other end, then retract the rod and the grease will be drawn back in. 

6.	With the "plunger rod" inside the bearing, and the grease plug being at the exact edge of the other end of the bearing, push another rod in from the other side. The idea here is to introduce as little air as possible in between the two rods. If the plug is too far inside then you'll have air and grease between the two rods.

7.	Compress both rods evenly until the grease starts squirting out the sides of the bearing. Do not apply excessive force. First the grease will be absorbed, then the transport oil will start bubbling out, then clean grease will come out.

8.	When both smooth rods touch within the bearing, observe if only clean grease was coming out. If by the time the rods touched there was still fluid then there's still oil in the bearing. It's possible that if you didn't use enough grease nothing would've come out.

9.	Slide the bearing up and down the smooth rod as to get the bearings accustomed to the new grease and to shed any excess. Pull the rod completely out and push it back in. Do this two or three times until there is no excess of grease clumping up on the bearing clearance.

10.	Clean the huge mess you just made.

This method is effective but has a couple of issues. First, it's a bit wasteful on grease which isn't a big deal, and it's also possible to push out grease and replace it with air voids, also not a big deal since the bearings will be greased and some of the later steps work to minimize this. I realize my procedure may not be perfectly clear so I'll try to make a video of it so you can certify I'm being accurate in my description.

dgcaste, 2016-04-10:
Video done!

https://youtu.be/N3Ec29160h0

mzbotreprap, 2016-04-11:
This is awesome! For a guy who's not actually building one of these, you have contributed quite a bit to the project :)

dgcaste, 2016-04-11:
Thanks man! I am genuinely excited for this project and wish I could contribute more. I am living vicariously while I put together the resources to grow into this build. Figured I'd let it mature a bit and learn more about the CoreXY by making AlexY's FuseBox, and I'll recycle motors and bearings so there's some synergy. Odds are I'll mod your box to run a single E3D v6 which is what I'm building with now. 

Regarding the current instruction, I feel like I want to tighten up the procedure to make it easier to read. I think the video works well though. Do you see any issue with this method? I don't want to blow a seal, haven't tried the Misumi lithium soap based grease to know if it's more or less viscous. I'm also noting that motion is not perfectly smooth and noiseless which I was expecting, in all my years of engineering I never dealt with a perfectly calibrated linear bushing installation.

porterble, 2016-04-15:
Just a quick note on this one, be careful with the force you exert on the rods when using them as plungers. If you push too vigorously you can pop out the end seals, which is fairly disastrous.

mzbotreprap, 2016-04-15:
This is good advice.

dgcaste, 2016-04-15:
I addressed this in one of the steps and I believe I did it in the video but could stand to be mentioned as a warning in the beginning since it's so critical.

porterble, 2016-04-15:
Ah, missed it in the step, and also didnt watch the video as i was on the train :) Probably still worth a warning for newcomers.

dgcaste, 2016-04-29:
Want me to close this? I don't know if you're considering adding some form of it to the manual, and if you are, if you're going to use the issue to remind yourself.

mzbotreprap, 2016-04-29:
No, let's keep it open. It's my to-do for updating the procedures in the manual.

commit 7d8a7db5 (2016-05-05): Fixes #26, fixes #22, resolves #11
closed 2016-05-05 by mzbotreprap