mzbotreprap/VORON #12: Frame and linear rod section lengths

porterble, 2016-04-15:
Just wanted to confirm that the frame lengths should be 370mm. From the imgur album i note that the spec list was a 390mm square footprint, but the 370mm length in the BOM plus two 20mm corner cubes would indicate a 410mm footprint. The diagonal distance on page 9 of the manual corroborates 370mm as well, but figured it would be good to confirm.

Same goes for the linear rods, the part code PSFJ8-320 would indicate 320mm, but want to make 100% sure before i get cutting. All that measure twice cut once stuff ;)

mzbotreprap, 2016-04-15:
Ah, yes, 20mm on each end do push the frame footprint beyond 400mm.

You are correct:
- Extrusions are all 370mm long
- Shafts are all 320mm long

porterble, 2016-04-15:
Thanks, sounds good. Although now im debating pushing it out another 25mm or 50mm as i have the lengths to do so.

mzbotreprap, 2016-04-15:
If you are cutting the stock yourself, and can source bigger build plates, absolutely do it. The 8mm shafts and 2020 extrusions will be ok up until 300mm (12") cubed build area. So, anything between the 9" (228mm) and 12" (305mm) is doable.

Just make sure the extrusions are 6000 series aluminum, and the rods are hardened. Otherwise you're asking for trouble, even at 9x9x9

porterble, 2016-04-15:
Yeah, according to the spec sheet from the provider it is 6061-T6 and the rods are hardened. Will be finding out soon how accurate the specs are i guess.

closed 2016-04-18 by mzbotreprap