mzbotreprap/VORON #13: Suggestion for Chimera carriage

porterble, 2016-04-17:
For those using non-official Chimeras or have to use 1/8" threaded bowden couplers due to the hole flogging out (thats me) then opening up the two filament feed holes on the carriage to 11.4mm would be beneficial. The majority of the threaded bowden couplers are 11mm dia on the push to disconnect piece.

mzbotreprap, 2016-04-17:
Ah, did not know this was a thing. I'll address it.

dcousens, 2016-04-18:
@mzbotreprap if you can,  prefer a parametric file,  maybe OpenSCAD,  such that you don't need to have hacks in the design to account for all things :+1:

porterble, 2016-04-18:
While i agree with parametric options for other variations. I think that opening up the carriage holes is a minimal change that will work perfectly with both options. In fact it will give extra room for manipulating the inset couplers.

dcousens, 2016-04-18:
@porterble oh definitely,  I mean,  just in general.

porterble, 2016-04-18:
@dcousens for sure. See my main suggestion for parametrisation  :)

mzbotreprap, 2016-04-18:
@porterble I like this. I actually have a perfect fit BSP threads modeled on the extruder already.

@dcousens Parametric all the things is another project in itself. Fusion360 doesn't play well with OpenSCAD, so no OpenSCAD exports unfortunately. I haven't tested if Fusion can actually export parametric STEP files, which you can then bring into something like FreeCAD and play with.

mzbotreprap, 2016-05-05:
Ok, finally got to this. Here's the solution: http://imgur.com/y8i2zFJ

Adding the BSP top half to the repo now

commit cbe5cca3 (2016-05-05): Fixes #13
closed 2016-05-05 by mzbotreprap