mzbotreprap/VORON #14: Parametrised PSU casing?

porterble, 2016-04-18:
Im guessing that the cutout in the bottom of the PSU casing is for the SSR to sit and be attached to the frame as a heatsink. The dimensions look correct for the OMRON SSR that you specify. However, for those of us outside of the US, and those who will likely use a DC heating bed then it could be good to parametrise the PSU casing. The likely SSR for DC usage comes in the same size as the Fotek SSR-40s which is 60x45mm. That also seems to be a defacto standard for SSRs in other applications too.

This would also be useful for different sized PSUs, my existing PSU i have realised is 100x50mm rather than 110x50mm.

Thanks for the great job you are doing! 👍 x 💯

mzbotreprap, 2016-04-18:
Are there no standards anymore? :) Parametrizing the design for the power box should be pretty simple. The rest of the bot is a different story...

Thanks!

dcousens, 2016-04-18:
There are plenty already out there, if that is suitable: https://www.thingiverse.com/search/page:2?q=psu+enclosure&sa=

porterble, 2016-04-18:
@dcousens yeah, i have designed a few on there. One of the things with the VORON enclosure though is that it includes the cutout for the SSR, which I am keen to keep. I would simply modify the existing STL, but then we are back in the territory of multiple versions floating everywhere, and for a project of this quality I figure its probably best to keep a strong 'known good' repository.

mzbotreprap, 2016-04-20:
@porterble Out of curiosity, how high are the mounting holes on your unit? Wonder if it's just taller, or is it a completely different PSU size/mount pattern I'm not familiar with. Do you have a model number for it?

porterble, 2016-05-04:
Ooops, forgot to link the imgur gallery over here: https://imgur.com/a/xNju6

The other one i have is just a completely different design, i have ordered another PSU from another supplier to try and get one that fits the spec...

mzbotreprap, 2016-07-18:
Do we still need this? Trying to clean things up a bit

lufton, 2017-03-20:
Yeah! I ran the same problem with my PSU. Mine has dimensions: 100x45x145+
Only mounting holes it has is on top/bottom and back side (see photo).

Here is some photos on [Flickr](https://www.flickr.com/photos/lufton/shares/0Qm9Z5).

closed 2018-05-20 by Maglin