mzbotreprap/VORON #16: STEP files

quixotic120, 2016-04-20:
hey

are you willing to share more step files? I wanted to edit the bed support carriage to work with a delrin anti-backlash leadscrew nut (like this http://qtecstore.com/pages/leadscrew_nuts/6mm_leadscrew_nuts/common/6mm_screwnutx2.jpg) because that's what I have on hand. will share it along with x midsection I'm making for the diamond hotend

thanks!

mzbotreprap, 2016-04-20:
Of course. I think I actually have this already desinged actually. This was one of the parts I was testing.

Lemme dog though the revision history

mzbotreprap, 2016-04-20:
s/dog/dig/

quixotic120, 2016-04-20:
Thanks! I don't follow though. Is that in the rev history somewhere? I don't see it

on the diamond hotend part - i'll open another issue bc I don't want this thread to get confusing

mzbotreprap, 2016-04-20:
Rev history is in my Fusion360 account :) There's a year worth of experiments in there, all the way back when I was just trying to design a better TAZ. Doesn't make sense to dump it all here though.

quixotic120, 2016-04-20:
Ah! do you have it shared? I can dig through if so, fusion360 would actually be great for me

mzbotreprap, 2016-04-23:
Added the STEP file for the bed carriage. I couldn't find the version for those blocks, sorry. In ether case, it would have been very out of date.

I don't have it shared, since it looks like a college dorm room. As a grown ass man, it's embarrassing to have that kind of mess exposed to the public :)

porterble, 2016-04-23:
an ass man... oh dear :p

quixotic120, 2016-04-25:
Understand not sharing the fusion360 acct - mine is also pretty awful.

Thanks for sharing the step though! Pretty busy this week but should have some time next week. If I get something working I will share it!

closed 2016-04-25 by quixotic120