mzbotreprap/VORON #17: PSU

gcormier, 2016-04-20:
I think BOM PSU source should be changed to OMC Online.

ie.

http://www.omc-stepperonline.com/switching-power-supply-200w-12v-165a-for-3d-printer-kits-115v230v-s20112-p-194.html

It's likely cheaper then AliExpress - I'm finding them between 15-25 and then at least 12$ shipping.

We need to obtain stepper motors from OMC anyways so it's already one of the stores to hit on the shopping list.

quixotic120, 2016-04-20:
also available on amazon for about the same price with cheap (or free w/ prime) shipping 

$19, 200W http://www.amazon.com/DMiotech®-DC12V-Supply-Switch-Converter/dp/B018TG7AXQ/ref=sr_1_38?s=pc&ie=UTF8&qid=1461173013&sr=1-38&keywords=12v+power+supply

$24, 360W http://www.amazon.com/eTopxizu-Universal-Regulated-Switching-Computer/dp/B00D7CWSCG?ie=UTF8&psc=1&redirect=true&ref_=oh_aui_detailpage_o02_s03

I have the second one powering a makerfarm printer and can attest that it does the job but it's a little overkill, probably worth saving the 5-8 bucks for a 200w

gcormier, 2016-04-20:
Very cool.

I wonder if there is an easier way to manage the BOM. A concise listing is
nice, but then a broken out listing with all these potential sources would
also be great... but as items are added to the master BOM we don't want to
rework the secondary list every time.

quixotic120, 2016-04-20:
the simplest solution would be to just add more columns to the spreadsheet (e.g. alt source 1, 2, etc after the main listing) but thats certainly not elegant

codexmas, 2016-04-20:
Even simpler, make your own copy of the BOM spreadsheet and modify as needed.
The BOM hasn't really had a lot of significant changes since 1.01, I used it as a checklist to coordinate things I already had vs ones I needed to order. I also found some of the items available at multiple locations and was able to consolidate orders.

I added columns like these to my copy:
Stock - Number of how many I already have on hand
Received - Tracking orders placed, and that they arrived, easy to watch outstanding deliveries
OnOrder - Number of items ordered from source
From - Actual source of supplier I ordered from
Bought - Order for item actually placed and paid for
Remaining - Calculation of Stock value against OnOrder
Cost - Actual costs of items I found online
Subtotal - Total of item cost and Quantity field
Saved - Calculation of subtotal of items I had already

Several of these columns then had sum fields at the bottom, showing I saved almost $300 in parts that were lying around.

So, I suggest that the more useful and interesting efforts for this project are 3D model updates, Documentation of the process and of course Videos :)

closed 2016-10-15 by gcormier