mzbotreprap/VORON #18: wider hotends

quixotic120, 2016-04-20:
Hey

I'm working on adapting the diamond color mixing hotend to the x carriage but I fear it may need a more significant revision. It appears that it is too wide to fit between the 8mm rails, may need to widen the XY joints and the carriage ends as a result. However, I have a concern re: widening the XY joints as they appear to fit the LM8LUU bearing fairly snugly and widening would either compromise that or force the top half to be molded outwards from the bearing which I believe would compromise the integrity of the part. I will admit that I haven't spent a ton of time on this, there very well may be a simpler solution that I've overlooked. 

original mounts available @ http://www.thingiverse.com/thing:720520 if anyone wants to take a look.

mzbotreprap, 2016-04-20:
Oh, holy hell, I forgot about this monstrosity. I think this can be mounted in the existing architecture. You are gonna lose some gantry travel if you start widening things.

Have you actually printed with this thing? How do you clean it in case of jamming issues?

quixotic120, 2016-04-20:
I have and it works well once you get everything setup; honestly the software end is the harder part to set up IMO. However, once it's setup it's pretty addictive. Printing in color is cool and experimenting with mixing is interesting

I haven't had a major jam yet (thankfully) but I have done successful cold pulls on it. Luckily the nozzle is fairly inexpensive (25 bucks or so) so worst case I would clean the e3d heatsinks as normal and just replace the nozzle. I suppose one could try clearing it with a drill bit but I'm always very hesitant to do that on a brass nozzle

I didn't even think about travel distance though, good point. It very well may fit though, I haven't tried much at this point and just wanted to throw out the idea for input. I'm printing the x carriage parts now to play around with the mounts I have printed already. Looking over the scad and the step files I think I have an idea on how to do it

Fheilmann, 2016-04-22:
I have similar thoughts about the two E3D v6 that I own. I don't want to go out and buy a chimera, but I believe wanting to fit 2 E3Ds side by side will become tricky.

dgcaste, 2016-04-25:
@quixotic120 -- I'd be delighted if you could take a few to review the ins and outs of this hotted. The idea is sexy as hell. 

@Fheilmann -- I have an e3d v6, I don't see an issue, especially if they share one fan exhaust

mzbotreprap, 2016-04-25:
Mounting dual E3D v6's is doable, but you will lose 12.5mm of X travel, and will need 55mm carriage bolts.

Those heatsinks are [pretty chunky](http://imgur.com/m1LmcjD)

Alternatively, you can mount them fully below the carriage, and sacrifice a little Z travel

dgcaste, 2016-04-25:
Now that you put them in a reference they are, compared to the carriage. Personally I don't mind sacrificing Z, I've found that most prints I make are 80% X-Y and I've never pushed beyond 100mm on Z. That doesn't mean that others won't :)

Fheilmann, 2016-04-26:
I guess I'll start with moving a single E3D v6 over from my prusa for now. 

The problem I have is that my mounts will probably have to be height adjustible, since one of my v6s - for some reason, is about half a milimeter taller than the other one :/

I'll try to contribute what I have once I get to it.

mzbotreprap, 2016-04-26:
When you're mounting two separate hotends, having them be height adjustable is pretty important. Luckily, you can just add room for them to play up and down, and let the carriage bolts do the job of holding them in place once you got them leveled. Of you can go with the fancy set screw design.

quixotic120, 2016-04-26:
@dgcaste super brief:

ins - color mixing is awesome and it does a fairly good job of it

outs - size/weight, potential for nasty jams is there I guess. Biggest out for me right now is lack of a key channel approximation - with three inputs I can do CMY but this eliminates K (black). Also, CMYK is subtractive and based on white paper so for true full color you would need a 5 channel hotend (CMY+black+white). There are people working on such a thing but I have no idea how feasible it is for retail purchase anytime soon. The only other con is adding color to models - kind of a pain in the ass right now but I'd imagine if color 3d printing becomes more mainstream this will be addressed more. The only other caveat I can think of is the need for 3 extruders but depending on what electronics you use this may or may not be a big deal. It's fairly simple to add an extruder to RAMPS and some of the more modern electronics come ready with 3+ extruder support. If we get to 5 extruders that may be a bit more of a headache and I would imagine this is where you may start seeing the limits of 8 bit hardware

https://www.youtube.com/watch?v=3Z_21b3xl20 is a good overview as well.

quixotic120, 2016-05-13:
just as an update - I have a potential bracket designed. I've printed it but haven't tested yet. Building my Voron sometime next week (finally have everything except for the heater). if it works I'll post it. It's big though so you do lose a bit of z height.

closed 2018-05-20 by Maglin