mzbotreprap/VORON #19: X Carriage Bearing length for single extruder setup

Fheilmann, 2016-04-28:
Hi there,

I have a question about the X-carriage in single-extruder configuration. I printed the necessary parts, and it seems that, with only one extruder, the double length LM8LUU should be replaced by single length LM8UUs.

Is this correct? If so the BOM and manual don't reflect these things, yet.

Bests

Florian

mzbotreprap, 2016-04-28:
Hey,

The X Carriage uses single heights LM8UUs regardless of the hot end configuration. That's the 2 in the BOM. The rest are used in the XY Joints and the Bed Carriages. You can check the manual for reference to what I'm talking about.

Fheilmann, 2016-04-28:
Shoot. Sorry I completely missed that. Closed.

closed 2016-04-28 by Fheilmann