mzbotreprap/VORON #20: XY joint mod

dgcaste, 2016-04-28:
Hey Maks, I had a crazy idea that I wanted to share with you.

The way you've designed the XY joints is to have bolts come in from the sides to close in and tighten on the smooth rods. This introduces two issues:

1- Repairability of the system is lower as it complicates partial disassembly for things such as bearing maintenance. You'd have to take the extruder apart in order to separate those two smooth rods out of their place.
2- Much less significantly, it may take away from the rigidity of the system by having a joint perpendicular to the direction of stress and motion.

I've attached a file for what I think to be a solution, which is to make the XY joint made into a top half and a lower half. Taking off the carriage and XY smooth rods would be a cinch, and probably simplifies servicing the gantry rods as well. It does elongate the XY joint a tiny bit though to make space for the vertical fasteners. Check it out and let me know what you think. I'm probably missing a few considerations that you've already taken over the last year :)

Daniel
[image]

mzbotreprap, 2016-04-29:
There are a few things to consider here. As with everything, design is a give and take process.

First, let me address the 2 driving motivations behind the mod:  

1. Getting to the bearings is actually pretty easy. Just loosen all retainer screws on both XY Joint, and slide the shafts out of one side. Then tilt it up, and slide it out the other. You can now get to the X bearings without any issues. Y bearings are even easier, as you can simply slide the Y shafts forward and slide the whole X axis off them. I've actually done this when I was re-packing them couple of days ago.
2. The way the shafts are retained now provides 2 things: It clams down the X shafts and applies the most force from the side, which prevents the XY wobble. The axis is a slightly less rigid in up/down direction, as it relies on the channels to keep it in place. You're proposing to do the opposite, which will actually introduce wobble in the directions we don't want wobble. The other thing it does is clamps down on the Y bearings,solidifying whole assembly. Basically the sides of XY joints pull double duty.

Then there's printability consider. I've had a few one-piece XY Join versions, and pretty much all of them failed around bearing retainers. With this parts, you have a limited number of ways to print it while avoiding supports, and relying on layer adhesion. Because of this you'll have to add plastic not just for the retaining bolts (which will in fact reduce your Y travel) but also for the Y bearing retainer, which effects all other parts it's flying past. That space around those parts is pretty crowded if you look down the Y shaft :)

Please let me know if I misunderstood any of what you proposed.

Keep the crazy ideas coming! It's a learning experience for everybody.

dgcaste, 2016-04-29:
This all makes sense. I knew I missed something(s)! Thanks for taking the time to explain.

dcousens, 2016-04-29:
I agree with this change :+1:

closed 2016-04-29 by dgcaste