mzbotreprap/VORON #21: Documentation: Missing endstop

codexmas, 2016-05-04:
For the A/B motor mount assembly part there is no indication of the endstop being added like the XY carriage has. Easy enough to finger out, but would be nice to have.

porterble, 2016-05-04:
I would presume this would come in the wiring stage. The main reason the end stop on the XY carriage is there is that its an absolute pain to fit later on once the carriage is in position with the extruder.

codexmas, 2016-05-04:
True, though I think it might be easier to add before mounting the motor to the frame. This is really a very small thing, just running through the build now and noting anything that might be worth adding.

porterble, 2016-05-04:
Its actually far easier to add afterwards. You have super clear access to the mounting point, and you also don't have to deal with the endstop pigtails when you are aligning the motor mounts.

I originally mounted it when building the motor mounts and then pulled it off for assembly.

mzbotreprap, 2016-05-04:
Ha! I knew I left something out. I have that illustration, but forgot to include that step.

And @porterble is on the money about X stop being a pain to mount once everything is assembled.

dcousens, 2016-05-04:
@mzbotreprap I think @porterble was saying it was a pain **until** everything is assembled, then it is easier?

mzbotreprap, 2016-05-04:
As someone who has removed and reinstalled X endstop about 30 times, I guarantee you it's easier to screw it in place while the carriage end is off the shafts, or at least not bolted down to the rest of the X carriage. The latter is not bad actually, so long as you drop the bed. It's still serviceable, so don't fret :)

porterble, 2016-05-04:
No as above, the X end stop is much easier to install when its off the shafts, especially for the first time with the tiny self tapping screws. Way harder to install once its been assembled (as i did after changing from the Rev A to Rev B carriage ends)

The Y axis stop is the reverse though, much easier to mount to the motor bracket after the bracket is installed on the frame. Especially if you are using hammer head T-nuts, mainly because its one less item to get in the way, or more specifically the cables getting in the way.

codexmas, 2016-05-04:
Sounds good, perhaps a mention in the Y axis assembly that the endstop is mounted later.

Note that I also followed the same pigtail design for the Y axis endstop, my wiring harness can stay in place if I need to pull the carriage or mounts.

mzbotreprap, 2016-05-05:
@codexmas Ok, I knew I haven't lost my mind. Check page 54 of the manual :) It's there.

closed 2016-05-05 by mzbotreprap