mzbotreprap/VORON #22: Documentation: Greasing bearings

codexmas, 2016-05-04:
As I am running through the build, I noted that the first occurrence of a bearing needing to be greased was on page 20. I would like to suggest that the greasing instructions be moved earlier in the docs into a preparatory section.

The flow once you start building would be much improved, as you get all the dirty messy bits done at once and not have to take a break to do them in the middle of assembly.

mzbotreprap, 2016-05-04:
I like this. Are you in QA by any chance? Cause this is some quality QA 🤓

porterble, 2016-05-04:
Would make sense to go with the section A.3. When i built mine i greased them all up at the start.

codexmas, 2016-05-04:
Not in QA, but certainly a programmer. Having a ton of fun doing the mechanical and enjoying the process.

commit 7d8a7db5 (2016-05-05): Fixes #26, fixes #22, resolves #11
closed 2016-05-05 by mzbotreprap