mzbotreprap/VORON #23: Documentation: Multiplication fail

codexmas, 2016-05-04:
This is really my issue, but in all the excitement to print things out I never really thought about some of the part counts in the folders, and that the Z Tower folder also specifies (2x)

So, I will be printing another set of the upper and lower braces cause dumb :)

Maybe change the folder name to be just Z Tower and the part counts to be (4x)?

porterble, 2016-05-04:
Haha, i did the same thing earlier, and so ill take the rap for having the x2 in both places, see an earlier closed ticket. Made sense to me at the time, but in hindsight the x4 notation would be more obvious.

mzbotreprap, 2016-05-04:
I think I overthought this and went too inductive on the folders. Need to pull back into reality and flatten things out.

dcousens, 2016-05-04:
@mzbotreprap that would be much preferred IMHO :+1:

timmit99, 2016-05-04:
Maybe make a PDF of what printed parts are required with a rendering next to the name and the amounts needed.

commit 8128a64a (2016-06-07): CLEAN ALL THE THINGS. Fixes #30, Fixes #23
closed 2016-06-07 by mzbotreprap