mzbotreprap/VORON #24: Documentation: Tnutz

codexmas, 2016-05-05:
While building, after assembling the Z axis and Gantry, I would like suggest there be some brief directions regarding the position of the T-Nuts. And/or explicit instructions to leave assemblies till later.

Essentially *some* people might get excited and mount the Z-Axis before the Gantry, and then realize that one of the T-Nuts is trapped at the back...

It is of course obvious looking at the diagrams where they all go and the purpose, but that is a pretty high cognitive load when you are assembling something like this. Pretty much like tunnel vision, especially the first time.

mzbotreprap, 2016-05-05:
Oh boy... I've been in this situation before :) Having to disassemble things sucks.

I'll throw a caution into the manual.

quixotic120, 2016-05-13:
a bit more money but drop in t-nuts are amazing for this reason

http://www.amazon.com/dp/B00WW6RE1I/ref=pd_lpo_sbs_dp_ss_2?pf_rd_p=1944687762&pf_rd_s=lpo-top-stripe-1&pf_rd_t=201&pf_rd_i=B00TG0QVWG&pf_rd_m=ATVPDKIKX0DER&pf_rd_r=1FSR7ZHBETYXEDBKB7G1 - these are really expensive but there are cheaper versions out there. These are just an example of what I mean

timmit99, 2016-05-21:
I don't know if i'm being an idiot here or what but for the upper left and upper right rails shouldn't the bottom slot have 4 T-Nuts? 1 for the motor mount, 2 for the upper z brace and 1 for the idler mount? I'm working through assembly and i followed the manual but i have 2 t-nuts on the outside of the upper left and right rails with nothing to attach them too and am missing 2 on the bottom slot for the z-brace.

mzbotreprap, 2016-05-21:
<insert explitive here> The T-nuts don't add up to 36!

Yes, you need 4 on the bottom of each side upper, and only 1 on the outside for the spool holders. I'll fix this tonight. 

This is why I'm only shipping out hammer heads now :(

commit bb55f90c (2016-05-21): Fixes #25, Fixes #24
closed 2016-05-21 by mzbotreprap