mzbotreprap/VORON #25: Documentation: Belts

codexmas, 2016-05-05:
Wanted to suggest that an approximate belt length be specified in the documentation.
The BOM only lists 3 meters, which once can assume means 1.5/belt.
I had a roll of GT2 already and cut quite a bit of extra length just incase.

mzbotreprap, 2016-05-05:
This is an excellent idea

closed 2016-05-21 by mzbotreprap

dcousens, 2016-05-21:
@codexmas careful,  the belts not be of equal length IIRC

mzbotreprap, 2016-05-21:
@dcousens how did you end up with belts that are not of equal length?

dcousens, 2016-05-21:
@mzbotreprap you know,  that is a good question.   I'm basing this purely on the measurements from OpenSCAD which gives me;

```
ECHO: "GT2_path length:", 1371
ECHO: "GT2_path length:", 1375
```

I wonder where that 4mm comes from :smiley:

commit bb55f90c (2016-05-21): Fixes #25, Fixes #24