mzbotreprap/VORON #26: Documentation: T-Nut counts

codexmas, 2016-05-05:
Possibly found that there are two missing T-Nuts from the Upper Front.
It appears I am two short for mounting the idlers on the top of the rails?

codexmas, 2016-05-05:
I can steal two from the left front vertical for what I presume to be the one of the cold end mounts?

mzbotreprap, 2016-05-05:
This is precisely why I have a giant bag of hammer head post-insert T-nuts coming in tomorrow. The pre-inserts are a nightmare to fix later on.

Let me check the manual.

mzbotreprap, 2016-05-05:
Yup, this is a mess up on my part. Many apologies. Here's the fix:

Upper Left and Upper Right only need 1 T-nut each on the top slot. The remaining 2 T-Nuts need to go into the Upper Front top slot.

I'm gonna update the manual right now (need to indicate the change of the RAMPs box to the back)

codexmas, 2016-05-05:
Cool, glad to have caught this early. It only took a few minutes to steal the nuts from the vertical.

mzbotreprap, 2016-05-05:
Yes, very fortunate. Thank you.

Cold ends don't really need both M5 screws to keep them in place. There's almost no torque on them, so you should be ok. I would screw in an M5 screw into each of the 2 floating T nuts on the top left and right, so they don't rattle around.

codexmas, 2016-05-05:
I think what happened is the extruders were mounted at the front originally.
They are now at the back in the new layout. There are no nuts in the rear verticals except for one for the Ramps case.
That means there is also one missing for the left rear vertical for the upper mount of the power supply.

porterble, 2016-05-05:
Shouldnt there be two in the rear verticals for the RAMPS case? Or is the design such that it is held in by just one (as mine is because the button head interferes with the LCD expansion board).

porterble, 2016-05-05:
Ultimately though hammerhead nuts are the go. I have figured out a way to keep them aligned too, just use a finger nail or a small screwdriver to pull on the screw head and put some pressure on the nut while you are adjusting them.

mzbotreprap, 2016-05-05:
@codexmas There's actually quite a shuffle with T nuts. The lower right one for the PSU box needs to move to the left, and the verticals T-nuts needs to be re-shuffled. I'm a moron and lost the vector files for those illustrations, so I need to re-draw them

@porterble Yup, RAMPs box need 2 on ether end (unless you have V1 box, which only has holes on one side). If you are using the low profile cap screws, the LCD adapter don't interfere once the screws are in.

And yes, hammer heads is where it's at. Starting tomorrow, hardware kits are gonna include those instead. @codexmas l'm gonna ship you some of those. Email, or PM me on reddit as to how many you need.

mzbotreprap, 2016-05-05:
I do want to keep the pre-assembly T-nut diagrams straight, to keep people's options open.

commit 7d8a7db5 (2016-05-05): Fixes #26, fixes #22, resolves #11
closed 2016-05-05 by mzbotreprap