mzbotreprap/VORON #27: lead screw stabilization

quixotic120, 2016-05-13:
do you have any issues with the top half of the lead screw being free to move? e.g. any unintended axial movement or wobble? I assume the linear shafts correct this somewhat but was wondering what you've seen practically.

mzbotreprap, 2016-05-13:
I had the screws contained at the top originally, which lead (no pun intended) to Z artifacts. You have to spend a lot more on the lead screws to have them be perfect. Removing the top constraints improved the print quality dramatically.

quixotic120, 2016-05-14:
ah that makes sense. I may add something to mine to try it out - I'm using lead screws salvaged from an old CNC so they are pretty high quality. However, they're coupled to steppers so that may eliminate any benefit of quality. Thanks for the insight!

closed 2016-05-14 by quixotic120