mzbotreprap/VORON #30: "Z Sensor Mount : Wire Anchor [REV A].stp" filename causes issues on some platforms

nallar, 2016-05-24:
Can't create a file with `:` in the name on Windows, so working with the repository on a windows machine is a pain.

dcousens, 2016-05-25:
Probably should just avoid file names with spaces and special characters

mzbotreprap, 2016-05-25:
This is gonna get fixed when I flatten out the STL folder. It's too fancy.

I'll go to a more OS agnostic file names.

closed 2016-06-07 by mzbotreprap

nallar, 2016-06-07:
That fixed the STL, but the STP still has a name with special characters: https://github.com/mzbotreprap/VORON/blob/416867d0a27b4f90c2b04ee9433b40fc8d030027/STEP/Gantry/X%20Carriage/Z%20Sensor%20Mount%20:%20Wire%20Anchor%20%5BREV%20A%5D.stp

mzbotreprap, 2016-06-07:
Wow, you are quick. That's coming next :)

mzbotreprap, 2016-06-07:
There you go. All clean and uniform. My inner OCD is getting better, I may even sleep now...

commit 8128a64a (2016-06-07): CLEAN ALL THE THINGS. Fixes #30, Fixes #23