mzbotreprap/VORON #35: Bed Support Carriage Nut Trap

mschweiner, 2016-06-08:
The nut traps on the Rev C bed support carriage are sized without any play, at exactly 5.50mm to match McMaster PN 90576A102. This creates significant difficulty in seating those parts if there is even minor overextrusion during printing. Other M3 nuts in the design can be self-seated by torquing down on the associated bolt; however, torques are much greater here given the locking nylon insert.

Given this additional required torque, it is likely that the hex shape in the ABS will strip if attempts are made to force the parts into place. Once this occurs, it is difficult or impossible to remove the nut as it is embedded within the stripped trap and further recessed in the carriage. This scenario really only plays out if a bolt is threaded all the way through in an attempt to seat the nut.

I'd recommend eliminating the nut trap mechanical feature in this instance, and/or specifying McMaster PN 91116A120 to be placed between the lock nut and the bed carriage. A 7/32" (5.56mm) hex socket can then be used to stabilize the 5.50mm hex nut from below while the M3 bolt is driven from above.

Might be a corner case, but I managed to play out the above scenario and ruin a bed carriage and two previously inserted LMUW8s upon trying to remove and salvage them.

mzbotreprap, 2016-06-08:
This is truly horrific, especially given the loss of the LMUs. :(

I've pondered the need for the lock nuts myself before, but never imagined this scenario. So, in order to not have this occur again, I'm proposing the following fix:

1. Replacing the lock nuts with regular M3 nuts, eliminating a BOM item in the process. 
2. Re-sizing the 3mm hole to be tighter when printed, and act as a locking mechanism for the screw
3. Re-sizing the M3 nut trap to allow for easier seating
4. Re-working the LMU seats to make insertion and removal easier

The nut trap feature is important as it's used when leveling the bed, and requiring another tool to hold the nut in place while the tension screw is tightened is not the more ideal user experience.

This should allow for self-seating of the nut, without the need to worry about destroying the nut trap. The spring, once tightened, should provide amble pressure to compensate for the loss of the locking M3 nut. I will also drop the nut seat down, to make seating it easier, and serve as a guidance to the user to tighten the bed springs properly.

I already have the solution for LMU seats, and will add the above mentioned improvements to the design.

mzbotreprap, 2016-06-08:
BTW, those nut traps where 5.4mm across the flats. They should have been 5.7mm as the rest of the parts to account for FDM process quirks.

I don't know where that dimension came from, but I'm fixing that as well.

mzbotreprap, 2016-06-09:
This is fixed with the new carriages. 

Commit [ba29dd4d08c8761f8e8054d40c507f8b980f67a5](https://github.com/mzbotreprap/VORON/commit/ba29dd4d08c8761f8e8054d40c507f8b980f67a5)

closed 2016-06-09 by mzbotreprap