mzbotreprap/VORON #40: Add a driver current adjustment guide

dgcaste, 2016-07-09:
Decent guides for tuning stepper driver current and voltage are very hard to find, and good ones are nonexistent. The project needs it and it might drive traffic to have the best guide on the internet, which currently wouldn't be too hard to achieve. Instructions on the manual with a linked video showing how is best.

mzbotreprap, 2016-07-18:
Yes, this is a good idea.

Are you volunteering to do the video? ;)

dgcaste, 2016-07-18:
I'd be more than happy to but if I made a guide it wouldn't be much better than anything that's out there as I'm not an expert. The idea is to have the definitive no-shit "this is for sure the best way to do this" method.

token47, 2017-01-31:
The worst part seems to always be the target voltage calculation based on the desired current. I made a pre-calculated table for that. Once you pass this step, it's easy. You have to 1) know your desired current, 2) see the table and set the voltage accordingly.

There are more complicated instructions for when you do not know your desired current, or you know but you would like to really tune it to your specific scenario, which involves trial and error, setting many voltages and seeing the result. You can map the interval from where the motors are too weak and skip steps to the point where they get too hot while operating and set it to the middle point on that interval. This is time consuming and requires patience.

There is also some important details like the A4988 ones (not sure about the DRV8825) will always give at maximum 70% of the configured current to the motors, even when you configure them to no micro-stepping (full step). This is intended and by design, but some people like to account for this and configure overcurrent so that 70% will be 100% on the motors.

The rule of thumb is, look at the table, configure the voltage and leave it alone if it works. If not, increase or decrease in small steps until it works and you are done.

[image]

Maglin, 2018-05-20:
This is very dependant on the driver and resister for driver ref used.  The user should source this info.

closed 2018-05-20 by Maglin