mzbotreprap/VORON #41: Differences in full graphical LCDs

dgcaste, 2016-07-14:
The LCDs that are bundled with the Amazon RAMPS boards (OSOYOO, Robotlinking, Kuman, etc) have slightly different dimensions that the ones the VORON model is configured for. For one reason or another I've had at least one of each of these, and several of the Robotlinking. It may have to do with me burning boards because I'm not paying attention to what I'm doing with conductive objects ;). Anyway, this physical difference results in several issues:

1. The two "middle" M3 holes on the board are about 1mm off where the plastic thinks they are, the M3 screws can be forced in but they don't line up perfectly. The fix would be to place the holes 1mm above where they are in the plastic
2. The metal encoder knob is longer so the knob shows metal under it even when it's fully inserted
3. The encoder knob doesn't have a "flat", instead it has a geared outside with a slot in the middle, making the VORON knob not fit altogether unless it's drilled out
4. There is a different clearance for the stop button, so when the unit is snugly assembled the button is effectively pushed all the time. The stop button groove on the case and the button itself need to be shortened
5. This is a high resolution encoder, so I had to change the pulses to 4 from 2 to get the encoder to work smoothly

mzbotreprap, 2016-07-16:
1. Can you provide a picture of which M3 holes need to move, and in which direction?
2 and 3. What's the dimensions of your encoder shaft?
4. Stop button, how much do I need to bring that in by?

I think we're gonna have to have different versions of this thing for different version of LCDs floating around. :(

mzbotreprap, 2016-07-17:
Never mind, I found an LCD that matches the one you have in the pile of parts. I see what you mean.

Adjusted design coming

mzbotreprap, 2016-07-18:
Fixed in the latest version

closed 2016-07-18 by mzbotreprap