mzbotreprap/VORON #50: Suggestion: Marlin UBL instead of Prusa Auto Mesh Leveling

Fheilmann, 2016-08-26:
I don't know how far along the Auto Mesh Leveling feature is for the voron, but I just discovered that someone is working on implementing a "Universal Bed Leveling Feature" for Marlin which is scheduled for inclusion in 1.1.1

The current code can be found in the official Marlin Repo under the `devel-ubl` branch. It is based on RC6 which I believe is also the current base for the voron. May be worth a look.

kmbecker13, 2016-08-26:
I'm curious about this - do you know what the differences are between UBL, ABL and MBL?  Seems like lots of flavors to choose from.

Fheilmann, 2016-08-26:
Automatic Bed Leveling (ABL) (as it is included in Marlin right now) probes a grid to determine if a bed is not level. It does not, however, account for situations where the bed is warped or otherwise irregular. ABL assumes that the bed is merely tilted and corrects accordingly.

The solution for such beds is to use Mesh Bed Leveling (MBL) where the bed is again probed in a grid. The difference here is that the correction is then also applied in a mesh, as MBL applies its corrections in a mesh. So if your bed has a "hill" or a "valley" or both, MBL can correct this better. The downside of MBL in its current state is that you have to perform it manually. The nozzle moves to the predefined grid points and you have to perform the "paper test" (lower the nozzle by hand until it grips a piece of paper) at each point. After you're done, the mesh can be stored in the EEPROM and can be loaded before each print (another advantage over ABL which is performed before every print)

The Firmware of the prusa MK2 has an automatic mesh leveling feature where MBL is performed automatically using the bed probe. I believe this is the functionality that the creator of the voron is trying to port to standard Marlin. The prusa, while based on marlin, is still very specific to the prusa mk2 so porting is not easy.

UBL is an attempt by some Marlin devs to "update" the marlin bed leveling toolkit with the automatic mesh leveling while also adding lots of useful extras (e.g. perform automatic MBL and manually add the areas where you cannot use the bed probe, or manually correct specific mesh points). UBL also includes the G26 command, which prints a one-layer thick test pattern across your whole bed to test your mesh and spot inaccurate areas. You can then adjust those specific mesh points.

For details, I suggest you read the readme of the `devel-ubl` branch which can be found[ here](https://github.com/MarlinFirmware/Marlin/blob/devel-ubl/README.md)

mzbotreprap, 2016-08-26:
ABL is also not great at dealing with twists in the axis. Say one of the sides of your Y axis is lower on one end the the other by 0.1mm. It's close to impossible to detect, but will confuse ABL. It's not much of a difference, but when you're trying to lay down perfect first 0.1mm layers, you're gonna have a bad time.  

UBL is very interesting. Thank you for posting this. I highly recommend throwing a topic up in the subreddit as I know at least a couple of people there who'd be willing to jump on this and test it out.

kmbecker13, 2016-08-26:
Thanks, you're the man for providing such a great summary - i actually have a Prusa Mk2 and LOVE the bed leveling functionality, seems like a mesh level is required for the PCB heatbeds since they aren't very flat or stiff.

I am very curious to see what type of difference UBL/Mesh leveling would offer for something like VORON with a relatively flat bed that is also fairly stiff.  I'm going to dive into the code this weekend and see what chaos i can cause...

kmbecker13, 2016-08-28:
I got this working on my VORON, but i have a larger bed so you will need to tweak the configuration.h file to match your build area.

I dumped the Marlin folder on my github here: https://github.com/kmbecker13/VORON-UBL

brunobasto, 2017-02-06:
Hey guys, any updates on this? Also, could the VORON benefit form the self xyz calibration that the Prussa i3 mk2 features? http://www.prusaprinters.org/first-printer-to-automatically-correct-geometry-in-all-axes/

closed 2017-07-25 by FHeilmann