mzbotreprap/VORON #51: Locknut on extruder gear

gcormier, 2016-09-08:
I've had a few failed prints due to the bolt coming loose. I've hit it with some Loctite carefull as it Loctite generally doesn't play well with plastics.... this should be a locknut.

Scozz0, 2016-09-08:
Highly recommend going for the purple 'light' strength loctite/permatex.

kmbecker13, 2016-09-08:
I had a similar issue with the nut coming off and then overtightened it to the point that i damaged the bearings in the extruder, definitely need to watch out for that.

dgcaste, 2016-09-08:
Can confirm I've had similar issue, resolved with tightening but stresses bearing a bit. Loctite or a locknut here would make sense.

gcormier, 2016-10-15:
Closing. Nut is supposed to have a second one afterwards.

closed 2016-10-15 by gcormier