mzbotreprap/VORON #52: Extruder gear

gcormier, 2016-09-08:
Using 2 washers, my extruder gear is still not aligned with the driver motor. I feel like it could use 4 washers instead.

gcormier, 2016-09-08:
[image]

dgcaste, 2016-09-08:
You can remove the grub screws on the metal pulley attached to the motor and slide it up and down or flip it around.

gcormier, 2016-09-08:
I looked at both options... neither seemed to work well. Flipped around it
was then offset the other way and I was worried about rubbing. Offsetting
it on the drive shaft causes fitment issues - as it stands I have about 1mm
between the metal puller and the rear Z stabilizer.

dgcaste, 2016-09-08:
Maybe you have an abnormally short motor shaft that doesn't give you the proper degrees of freedom

mzbotreprap, 2016-09-08:
First off where is your locking nut on the other side? You'll need to drop the GT2 pulley on the motor deeper to line it up with the drive gear.
[image]

gcormier, 2016-09-08:
Well shit. That'll do it :)

Hmm, do you have the collet screws tightened on the flat part of the motor
or the round part? I put mine as low as it could go. I'll double check
tonight.

mzbotreprap, 2016-09-08:
It's sitting on the round part. Not really an issue due to the fact that there isn't any real torque on it. You'll grind though your filament before those 2 grub screws slip since it's 1/4 of the torque you'r putting on the MK8 gear.

gcormier, 2016-09-08:
Ok bingo on that one.

pg 47 should be reworded. "Step 14: Attach the drive pulley using the same
technique you used to attach
the drive pulleys on the A and B motors"

To state you don't have to worry about the flat part of the spindle.

mzbotreprap, 2016-09-08:
Good point. I'll use this issue for tracking the manual change. rev 1h, here we come

dgcaste, 2016-09-08:
The manual in git under documentation seems to be the old one

mzbotreprap, 2016-09-08:
Yup. The Kit manual is what's being discussed here (mzbot.us/voron/manual)

I'll update the self-source manual in Git to the new look once I finish shipping out the kits.

FHeilmann, 2016-09-16:
@mzbotreprap When you update the manual, please make sure that the parts that are unique to "self-sourcers" remain. Most prominent example here would be the diagrams for the pre-insertion T-nuts.

mzbotreprap, 2016-09-16:
@FHeilmann I have no intensions to abandon the self-source community. Also, I spent way too much time on those pre-insertion diagram to not include them :)

porterble, 2016-09-16:
I thought the preinsertion diagrams were deprecated anyway as there isn't any need for it with the hammerhead nuts?

FHeilmann, 2016-09-16:
@porterble There are some unfortunate souls (like me) with no access to MISUMI extrusions. My 2020s, for example, have a 5mm slot, meaning I either have to use pre-insertion nuts or use hammernuts with M4 threads. 

There are nuts which can be inserted later (in fact I have a bunch of those for my addons), but since many people may go down the pre-insertion route in the future it should be kept.

Printeronimo, 2016-09-30:
@FHeilmann  Could you please share which kind of Nuts can be inserted later? I have the same issue with my 5mm slot extrutions. So far, i have not found a possibility to add nuts without disassembling the whole frame.

FHeilmann, 2016-09-30:
The ones I used look like this:

http://www.ebay.de/itm/131647310693?_trksid=p2060353.m2749.l2649&var=431039409180&ssPageName=STRK%3AMEBIDX%3AIT

They are a bit tricky to insert (especially if you get the ones with that little centering bit on top), but it beats disassembling the frame!

closed 2017-04-02 by gcormier