mzbotreprap/VORON #62: Unable to disable LCD buzzer

CarnufexG, 2016-11-23:
The chirping of the LCD is driving me crazy. I looked in configuration.h, pins.h, pins_...VORON.h and cannot find the right switch to turn it off. Tom Sanladerer's video showed a way to do it but that doesn't work. We need to disable that buzzer click/chirp/tick.

mzbotreprap, 2016-11-23:
I'm out of town for the holidays so I can't help reproduce the issue, but my money is on EM interference over the long cables. 

Here's something to try: take some aluminum foil and wrap both ribbon cables. You should hear the noise disappear or at least go outside of audible range. The permanent fix is alimunum tape from Home Depot or any other hardware store. It's normally used for sealing HVAC ducts.

CarnufexG, 2016-11-23:
Thanks, but it is definitely a steady "heartbeat". It is at the same 
frequency as the clicks when turning the knob. It is not EM because I 
have moved the machine to several locations and on several AC circuits 
and it is steady as clockwork. It sounds like a clock tic... like a 
timer or something. I tried ferrite coils with no results. If I 
initialize RAMPS it will stop until initialization is complete then 
start clicking again once a second. I will record it and post it on 
reddit. Have a good holiday.

mzbotreprap, 2016-11-23:
It's not from the external sources, it's from the cables themselves. The signal that drives the LCD is pretty high frequency. The chirp you're hearing happens when the LCD is redrawn. I just went though my stash of videos, and noticed it in one of them.

Where did you route the LCD cables? I stuffed them into the bottom of the lower right extrusion. The slot is just big enough to fit them. 

The ultimate solution is to remove the speaker. It's an annoying little beeper. You should be able to clip it off, but it's pretty drastic. Try shielding the cables from each other first.

CarnufexG, 2016-11-23:
I routed the cables along the sides of the extrusions. I did some research and found others have had this problem too. There does not seem to be a solution so I made one myself. I stuck a sock in it (stuffed some electrical tape into the piezo element) and it silenced it real good :) Problem solved, although I went through the Marlin pin-outs and the LCD schematics (http://bigtree-tech.com) and could not find a reason for it to not be silenced through a pin definition. Blessed silence from my printer, aah.

mzbotreprap, 2016-11-24:
Yeah, I think we can close it. The reason the pin assignment had no effect is because the sound you are hearing is not triggered by Arduino intentionally. It's a byproduct of re-drawing the LCD. For some reason redraw signal is inducing a "chirp". The LCD module may be missing some noise filtering on the buzzer circuit. 

Silencing the buzzer mechanically works a treat :)

closed 2016-11-24 by mzbotreprap