mzbotreprap/VORON #63: G29, Level Bed, intermittently causing bed to crash into nozzles.

CarnufexG, 2016-11-29:
When a G29, Level bed, g-code command is issued the 9 point bed leveling routine appears to not sense the Z probe (in the same spot every time it happens) causing the bed to crash into the nozzles. It has happened on 3 occasions and the X carriage is always in the middle right (A motor side, middle row) when the bed hits. I've stopped using G29 and the LCD panel as often. It is a very intermittent problem and cannot be produced on demand. The bot functions fine otherwise.

dgcaste, 2016-11-29:
Do you g28 first?

Daniel

CarnufexG, 2016-11-29:
Yes, I always G28 before G29.

dgcaste, 2016-11-29:
Could be an electrical connection thing there. Move the carriage to that area, get the nozzle to touch the bed so the probe is unquestionably in range, then wiggle the wire harness and see if the light flickers.

CarnufexG, 2016-11-30:
I ended up removing and repositioning the probe... was making noise anyway (resonance). Will see if it hiccups again. Thanks for the suggestions.

CarnufexG, 2016-12-07:
Induction probe mount cracked.

closed 2016-12-07 by CarnufexG