mzbotreprap/VORON #79: Marlin configuration.h steps/mm wrong

radry, 2017-01-28:
In the configuration.h the steps/mm for X and Y are set to 80. When using the drivers in 1/16 mode, like recommended, the steps/mm have to be 160.

For coreXY the steps/mm calculated from the pitch and toothed gear (80 in this case) has to be doubled. I tried it myself and with 80 steps/mm, the axis move only half of what they should.

`#define DEFAULT_AXIS_STEPS_PER_UNIT { 160, 160, 400, 608 }`
 instead of 
`#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 400, 608 }`

Printeronimo, 2017-01-29:
Sure you are not using 1/32 microstepping?
I did try that once and had to change the steps to 160/mm. For 1/16 though, 80 steps work just fine for me.

radry, 2017-01-29:
The jumpers are set to 1/16. How can I verify if it's running on 1/16 or 1/32? Although as I wrote, every source claims that steps have to be doubled for corexy.

Fulg, 2017-03-01:
Surely you fixed this by now, but the issue remains opened.

- What drivers are you using?
- What are your MS jumpers set to?

For me 80 steps worked just fine in 1/16 mode.

radry, 2017-03-01:
It's fixed. I made a mistake. The jumpers for X,Y and extruder were set to 1/32 but I forgot about it and got confused because I left the Z steppers on 1/16.

closed 2017-03-01 by radry