mzbotreprap/VORON #80: Directions of the motors on the COREXY model

token47, 2017-01-31:
Hi!!

After suffering a bit with this I have a table for this. It may help others, not sure if this is on the manual anywhere. Anyway, there it goes:

[image]

FHeilmann, 2017-01-31:
This also depends on the direction you are homing to.

For example I'm using what you call "both motors inverted" 
Since the endstops are  on the side of the B-Motor for X and on the B-Motor mount for Y this lets me home to min endstops in both directions. I believe this is also the way RCF has set it up.

Not saying your drawings are wrong, just that there is more than one "right" ;)

token47, 2017-02-01:
Yeah, you are right. I was thinking of your case as "motors at the front", and that's the reason for the drawing at the left. But the point was to go from the current config to the desired config in just one, maybe two steps instrad of trial and error, because that stuff can be confusing sometimes.

FHeilmann, 2017-02-01:
The thing is that instructions for this part of the assembly are not only based on motor location, but also in which direction (MIN or MAX) you want to home towards.

Another small note, the lower row mentions "swap X and Y connectors" while it should read "swap A and B connectors"

token47, 2017-02-01:
Well, you could be homing to MAX positions and still have the config I sugested even on voron. But that's not the point, this table I made before building voron and is meant as a generic table to correct the directions, considering the one you are and the one you want. Just ignore the draw on the left (I'll probably remove it).

Yes, they are really A and B motors but on the RAMPS they are labeled X and Y on the connectors, and that is what it meant (I guess every board will call them X and Y). I guess I will put a note on that on the bottom.

token47, 2017-02-01:
Here is a revised one. Thank you for the comments, and please keep sending them.

[image]

closed 2018-05-20 by Maglin