mzbotreprap/VORON #83: homing problem

DeathmannX, 2017-02-11:
Hi Guys,

first of i got no Voron i've build a D-Bot http://www.thingiverse.com/thing:1001065 but the Firmware Config should be identical.

If i try to home a Axis it moves maybe 2-3 mm and thats it.
If i trigger a Endstop by Hand and try to home the Axis it moves again like nothing has changed...
I checked my Endstops and my configs but i cann't find the issue. 

That is how my config looks like:
http://pastebin.com/z5egctYD

Can you Guys help me?

mzbotreprap, 2017-02-12:
Move the stuff off switches so you know they are not triggered and run M119 command. Does it show "triggered" for any endstop? If so that's your problem. Invert endstop logic and you should be ok.

DeathmannX, 2017-02-12:
I did what you said and this is the Result:

The first moves are simple move commands the other it tryed to home...

https://youtu.be/9twYnP1kZ3Q

DeathmannX, 2017-02-12:
I noticed that i didn't selected the right pins and forget to comment out the pin settings for my Y axis... 🤣 

But one other Thing, why does the Printer not show me when i heat up my Bed or Hotend at whitch temparture it currently is?
I can only see it in the console dialog. 😞

closed 2018-05-20 by Maglin