mzbotreprap/VORON #85: Clearance issue with the BSP adapter and finalized V1.5 gantry Z actuator

Fulg, 2017-03-01:
I am pretty sure the BSP adapter was designed before the final Z actuator rocker, because there is no space for the adjustment screw (added with the last revision I believe).

In order to use it I had to trim the rocker a bit next to the adjustment screw, remove some plastic around the edges, and remove the raised middle Chimera screw mount so the screw does not interfere with the rocker movement.

I do not have STLs to share, this was done in-situ with a pair of side cutters :)

[image]
[image]

mzbotreprap, 2017-03-01:
Well, it was designed with the final gantry, but when I was making the final tweaks, the whole assembly was off. This is the result. Just checked interference on the model, and I see the same thing.

I'll tweak the adapter vs. messing with the actuator arm. Don't really want to mess with that part.

Fulg, 2017-03-01:
On my carriage this is worse because I used socket head screws. If I had used a button head screw for the middle Chimera screw (where I clipped the adapter) it would have also worked. 

All of my 20mm screws are socket head however and I didn't want to mix&match screw sizes for the hotend :)

closed 2017-03-30 by mzbotreprap

Fulg, 2017-07-23:
Are you sure this was fixed correctly? In the latest version of the V1.5 carriage assembly, the new BSP adapter still does not allow room for the middle Chimera screw, next to the rocker. I modified mine to get clearance for the third screw, before printing it this time, it ended up looking practically identical to the one I manually mutilated and photographed above.

[image]

commit 2737b066 (2017-03-30): Removed rear fastener and beefed up the componnet. Fixes #85