mzbotreprap/VORON #98: Power Box wiring gauge is inadequate

superjamie, 2017-05-15:
The design currently suggests to use 20G wire between AC and heatbed.

Assuming the stock 250W heater at 110V that's 2.3A

According to http://www.powerstream.com/Wire_Size.htm the size 20G is not sufficient to safely carry this amount of power.

It should be *at least* 18G and probably more like 14G to provide over a 100% safety margin.

mzbotreprap, 2017-05-15:
Those numbers are given for 1000ft. 

Use the calculator they provide and plug in the 20AWG, 2.3A, and 1ft of wire (standard on 9" bed). You end up with 0.04% voltage drop, which equates to fairly negligible power dissipation. You of course can upsize all the wiring, but for the purposes of safety, 20AWG is plenty.

superjamie, 2017-05-15:
Fair enough, thanks for the correction.

closed 2017-05-15 by superjamie