mzbotreprap/VORON #101: XY Gantry lower idler reinforcement

pyctam, 2017-06-16:
Was doing v1.5 gantry Installation following video on YouTube. When partially slided shafts in as on https://youtu.be/fm0BPvC2oas?t=1m30s and focused to put on X axes on one side, shaft on the other side by its own weigh broke lower idler. Paying attention to the other idler shown that it is cracked too which lead me to think that there is an issue with either:

1) I didn't pay enough attention to the technological process, so broke idelr because of my "sloppiness", or
2) printed parts had poor quality, or
3) idlers design is weak and they should be re-enforced either by making them thicker and/or making shaft retainers longer along the idlers surface and amount them using 4 bolts.

It seems that third option might reduce possibility of the first two.

[image]
[image]

pyctam, 2017-06-16:
based on suggestion from https://www.reddit.com/r/voroncorexy/comments/6hkoq3/has_someone_had_an_issue_with_the_v15_xy_gantry/

`I left the rods hanging while only supported by the (loose) retainers on the front. Then, inserted the shafts halfway, then insert the X carriage, then push back the Y shaft all the way in (using the carriage to align the shaft at the back). Then repeated that for the other side.`

Also from there `that the plastic did not break along the layer lines. ... this excludes poorly printed parts.`

Installation instructions, suggest to insert shafts half way first, but I probably pushed them more, so the length and weight of the shaft behaved like a lever and broke the idler.

So, if you think that lower idlers are rigid enough, I think that in that case there should be a warning with explanation for the first time assembler to show what might happen if you do it wrong.

Fulg, 2017-06-16:
Note that I actually said "**I think** this excludes poorly printed parts", I don't know for sure. Maybe this wasn't even ABS. Since your parts are already broken, perhaps you can confirm that they were printed with 3 outlines and 40% grid infill.

I have had no issue with this, and printed ABS is usually quite resistant to abuse (sometimes surprisingly so). As I said on Reddit, from the pictures this looks like the kind of failure you see with PLA parts.

I guess it is better that they failed now rather than later setup, or worse, during a print. ;)

Maglin, 2017-06-16:
I printed those and they are indeed ABS.  That is very strange.  I do believe there might have been 2 sets of those in the box.  It looks like the weight of the Gantry twisted the nuts which broke out through the ABS.

Fulg, 2017-06-16:
Of course one should not let the gantry hang from the loose retainers of the idlers, you have to hold everything in place until **both** shafts get to the motor mounts at the back. It seems like this is what happened.

pyctam, 2017-06-16:
@Maglin, no blame at your work 👍 . As I stated above, it seems that I broke them by `pushing shafts more than half length, so shaft behaved like a lever and broke the idler`.

If you have another printer ready to re-print them, it's actually not a big problem . But current design of the shaft retainer is narrow, so shaft can swing. If retainer was wider (from one edge of the idler to another or for the rest of the shaft length) and mounted with the 4 screws on each edge, I think (but not sure) that idler could survive.

_P.S.: I will check box again, but yesterday evening I seen only spare motor mounts and upper idlers only. But it is not a problem. I have colleague at work (but he prints PLA only), so will ask him to re-print them and use them until I will be able to print them in ABS with VORON. If it will not work, than I will ask you to help me :-)_

Maglin, 2018-05-20:
An old issue to do with User letting the weight of Y rods pull down on the Ider.

closed 2018-05-20 by Maglin