mzbotreprap/VORON #104: M5x10 screws are too long

Kriechi, 2017-09-01:
I just tried assembling my self-sourced VORON 1.5 - but had to stop early because the M5x10 screws to fix everything to the M5 T-nuts & 2020 profiles are too long. I can screw them tight until the screw hits the inside of the 2020 slot, but the printed part is still loose.

Either the printed parts are too thin, or the screws are too long. I'm going to get some M5x8mm, which should fit and provide a solid screw connection.

Is anybody else experiencing this issue?
Are there differences in the slot depth of the 2020 profiles? I got mine from SMT in Germany (6mm slot, S106B2020)

Kriechi, 2017-09-02:
Apparently the 2020 profile from SMT have a shallower slot.
[image]

Compared to the Misumi ones:
[image]

I will updated #102 to fix the BOM.

pakerfeldt, 2017-09-02:
You are pasting the wrong SMT image. The depth of that profile is the same as misumi and the screws fits just fine.

Kriechi, 2017-09-02:
Thanks! My mistake - I replaced the image above to avoid confusion.

Kriechi, 2017-09-02:
Since most parts have a 4mm wall, I would argue that even the original Misumi profiles + M5x10 are not optimal, because the barely fit (4mm wall + 2mm profile wall + 4mm slot depth = 10mm screw length length).

PR #102 (2017-11-27): fix frame items
As discussed with immutef on Slack.

The correct 2020 profile uses a 6mm slot and M6 in the front face, but the article numbers pointed to a 5mm / M5 profile. This should now be the same as listed in `BOM/voron.md`

closes #104
closed 2017-11-27 by mzbotreprap