mzbotreprap/VORON #121: Interest in using GitHub releases

jschneidereit, 2018-02-13:
I noticed that there weren't any commits associated with the [releases section](https://github.com/mzbotreprap/VORON/releases) in this repo. Does any such list exist somewhere, or does one typically look through the commit history on master?

Would there be any interest in this? I'd be willing to contribute!

Fulg, 2018-02-14:
Can you do a Release _after the fact_? There is interest because everyone making a V1.5 is confused with regards to what changed between both versions, which bits in V1 you can skip for V1.5.

V2 is easier since it essentially shares no parts with V1.

I can help you sort out the bits if you want since I built both versions, I just don't know how to make them into a Release. Some of the information is likely spread out on the subreddit as well.

jschneidereit, 2018-02-14:
Yeah - it's easy to make a release after the fact!

To do so you'd just have to hunt the appropriate commit down in [master](https://github.com/mzbotreprap/VORON/commits/master) then add a release pointing to that commit with whatever version you'd like specified.

Instructions since I can't actually contribute this due to github permissions:
 - For v1.0, find the commit in master that is associated with that version
 - go to [releases](https://github.com/mzbotreprap/VORON/releases) and click `Draft a new release`
 - Insert `1.0` in "tag version" and search for first few chars in sha that you identified.
 - Add a title and description if so desired.

Boom! Now your users can easily identify the different versions, and even branch off of them easily!

Fulg, 2018-02-20:
I am still unsure how this would work. For V1.0 it is easy, just tag that specific commit. V1.5 is made of a subset of V1 and some new parts, I don't understand how you can tag that after the fact in GitHub, maybe you know.

I don't have write access to the repo either, this would be up to @mzbotreprap to set up.

jschneidereit, 2018-02-21:
Ah I see. I suppose that the commits with `V1 + new parts` could be patch releases e.g.: `V1.5.1`?
You just tag by choosing a specific commit. 

Some more information about versioning: https://semver.org/

Fulg, 2018-02-21:
I am familiar with semantic versioning, but the project history doesn't really fit this model, which is why it hasn't been done until now. It is not like a software product where new releases replace the old ones, new parts were just added to the pile we already had.

Still, it might be possible to make a new project branch that only include single version parts (replacing, deleting and adding as needed), and tag _that_ as a release.

cwhowell, 2018-02-22:
I think that approach would be a great improvement - I know I had to spend a bunch of time sorting through what I needed to print for V1.5, piecing it together from the directories in the repo and some outlines people made and posted on the subreddit.

Maglin, 2018-05-19:
I'm going to be working on making Voron Releases starting with V1.0 in the coming weeks as I have time.  Be patient with us.  Thank you.

shinuza, 2018-06-04:
@Maglin  If you need help with this, I'd happy to assist.

Maglin, 2018-06-05:
@shinuza thank you for your offer but I found the time tonight to take care of this.  Should make it easier for people going forward to build a V1.0 or V1.5 now.

closed 2018-06-05 by Maglin