mzbotreprap/VORON #123: VORON2: Insufficient hole spacing for Chinese spring nuts on some parts

Fulg, 2018-02-20:
For my VORON 2 I sourced some post-assembly spring nuts from AliExpress, since I didn't want to pay the premium for Misumi nuts. This exposed an interesting problem that is easily fixed at the source...

The chinese spring nuts are of varying lengths, between 13.8 and 14.3mm each. This means the closest hole spacing you can use is about 14mm. The hole is also far more offset from the center than genuine Misumi nuts, this is important later.

On the front idlers and A/B drives, the spacing for the holes are quite close (about 13mm). For these specific parts you cannot simply reverse one of the nuts to get the correct alignment, because then the extra bit is hanging outside the extrusion and interfering with the rail.

The hole spacing makes it impossible to use standard pre-assembly T-nuts, so you either have to use Misumi spring nuts or the rotating post-assembly nuts. I tried using the latter since I had some on hand but quickly got tired of fishing them out of my frame, and they have a tendency not to turn when making constant adjustments (which is required for running the Z belts).

I tried using one spring nut and one pre-assembly nut, this gives me the correct hole spacing for the stock parts, but still leaves a bit outside the extrusion that interferes with the Z rails.

This would be easily solved by extending the extrusion hole spacing by 2mm on the front idlers and A/B drives. In the end I simply elongated the holes on my already printed parts, which essentially accomplished the same thing, but changing the hole spacing would prevent this issue for future builders.

Let me know if you need more information from me. I don't consider this a show stopper at all but it is definitely a trap when using non-Misumi hardware, and it cost me a few extra hours of build time...

[image]

[image]

Fulg, 2018-03-01:
Turns out I was lucky with my 14mm Chinesium nuts, @pnokel on Slack made me realize that the equivalent Misumi nuts (HNTP5) are a whopping 17mm wide, which will definitely not work.

Closing, this is not an issue if you use the correct hardware...

closed 2018-03-01 by Fulg

PaulNokel, 2018-03-01:
I used an angle grinder to cut the HNTP5 into the required length.