mzbotreprap/VORON #128: Duet Bracket Hole Spacing

redblueboy, 2018-03-25:
Regarding the Duet Bracket STL in the STLs/VORON2/Base Compartment folder

The hole spacing is currently 2mm too wide. Holes should be spaced 115mm on center. Current spacing in STL is 117mm on center.

Duet Mounting Guide:
[https://duet3d.dozuki.com/Wiki/Mounting_and_cooling_the_board](url)

Duet Bracket STL:
[https://github.com/mzbotreprap/VORON/blob/master/STLs/VORON2/Base%20Compartment/Duet%20Bracket%20-%20From%202020%20Extrusion%20x4%20(Use%20Metal%20Standoffs%20or%20Plastic%20Spacers).STL](url)

Maglin, 2018-05-19:
Everyone with a Duet including myself has no issues with it.  2mm is a pretty big gap.

Maglin, 2018-05-21:
The issue has been verified.  So far seems links to the Duet Clone boards.  Others might need to chime in.  Truckmonky is working on a possible fix.

ubernero, 2018-05-26:
Tested printed bracket with my Fystec Duetwifi clone, no issues using motherboard standoffs

Maglin, 2018-11-10:
Cleaning house.  This issue has been resolved.  Thank you.

closed 2018-11-10 by Maglin