mzbotreprap/VORON #153: [Website request] Add links to the documentation

tobymurray, 2018-10-11:
Not sure if this is the right place, and may just be a WIP so feel free to close outright. I learned about Voron from Reddit and thought I'd check it out. The old configurator links don't work, as it looks like you've migrated sites. If I go to http://vorondesign.com/ right now, there's nothing really there. This is a non-trivial hurdle if I'm looking at a handful of designs and want to get some sense of what's going on. So currently I find http://vorondesign.com, go to GitHub, go to releases, find the link to 2.1, and go back to http://vorondesign.com/voron2.1 - feels a bit circuitous!

Awesome that you're using GitHub releases, consider adding a link to your landing page to make it discoverable that there is in fact stuff available (or even redirecting the landing page to the latest release?) Finding http://vorondesign.com/voron2.1 is presently more difficult than ideal - it looks like an awesome project and I think it'd make it more accessible to those of us who are coming from the "what's a voron" end of the spectrum.

New site looks great! Voron 2 looks super compelling.

tobymurray, 2018-10-11:
Must be brand new - I see the branding on the top of this repo still points to http://mzbot.us/. Maybe I just caught you right in the middle of the migration then.

mzbotreprap, 2018-10-11:
Yup. this is all a couple of weeks old. I'm migrating.

The reason you don't have VORON2.1 links off the main page is that i's not officially released yet. The manual is still in progress.

This is closer to what it'll look like when it's baked: http://vorondesign.com/voron2.1

tobymurray, 2018-10-12:
No worries - everything is looking sharp. I just learned about the Voron and was finding it a bit hard to get oriented from knowing absolutely nothing. I hope to be able to give 2.1 a shot!

closed 2018-10-12 by tobymurray