mzbotreprap/VORON #157: Mobius 2 source files?

dragonnn, 2018-11-05:
Hi,
any chance to get source files for Mobius 2? I only found Fusion360 files for the first version.

Maglin, 2018-11-10:
They are not released as of yet.  They will be released sometime in the future.

closed 2018-11-10 by Maglin

dragonnn, 2018-11-11:
Ok, I will wait impatiently! It looks like that perfect extruder, should not have problem with linear advance like most geared extuder with normal gears.

timmit99, 2018-11-11:
The STL's can be pretty easily modified in F360 if you are really itching to modify it.

numanair, 2019-01-03:
I would really like at least some STEP files for it.

drasrx, 2019-01-03:
> I would really like at least some STEP files for it.

Since this request was closed in November with the note that they will be released at some point in the future, it stands to reason that the point in the future where they will be released has not been reached yet.

The files will be released sometime in the future.

timmit99, 2019-01-03:
@numanair  which file are you wanting to modify? If its the base plate just make an adaptor. Fusion can give you the hole spacings and dimensions or you can even just remake it through an STL...

rothnic, 2019-01-21:
@numanair, I had the same need (I'm assuming) as you to modify the base. I didn't want to mess with the stl, so I created [some adapters](https://www.thingiverse.com/thing:3129044) for now. The orientation of the base is just likely to not be the best for most printers, because you'd need a longer bowden tube than necessary.