mzbotreprap/VORON #191: V2.1: Z-drive mount - prevent rotation of the extrusion

schlnk, 2019-02-04:
Due to design of Z-drive mounts rotation of the extrusion is possible. As a temporary workaround printed [L-brackets](https://discordapp.com/channels/460117602945990666/460142343014842371/536639687536934940) were used. 

Design of Z-drive mounts should be refactored in order to avoid potential rotation of the extrusion. 

Discord discussion is [here](https://discordapp.com/channels/460117602945990666/472450547534921729/534094280281554956).

[image]

pecirep, 2019-02-04:
While I agree this that this is a valid concern, I have to say 2 things:
1. The recommendation wasn't printed brackets, but internal L-brackets like [these](https://www.aliexpress.com/item/secret-link-why-are-you-reading-this-anyway/32898393769.html) (don't buy these, I have not tested them, it's the first source I found)
2. It has already been confirmed that 2.2 this won't be a problem in 2.2, only a few messages after your L brackets reference

schlnk, 2019-02-07:
z_rail_stops cant be used with L-brackets unfortunately. That's why printed brackets were designed. Good to know that It will be fixed soon, bug is just to ensure that.

Maglin, 2019-04-05:
This has been addressed.  Thank you.

closed 2019-04-05 by Maglin